alengwenus/ha-sma-ev-charger #1: Current development status

nanosonde, 2023-11-27:
Hi Andre!

Very nice that you are are working on a native integration for the SMA EV Chargers.

What is the current state of your custom integration?
Do you also plan to integrate mode switching as far as possible?

alengwenus, 2023-11-28:
Hi @nanosonde ,
it‘s basically done and already implemented in [this branch](https://github.com/alengwenus/home-assistant-core/tree/dev-smaev). I also wrote the underlying Python library [here](https://github.com/alengwenus/pysmaev/tree/dev).
I just need some time to make a custom component out of it (or integrate it into HACS), write some documentation and publish the Python package. 🙄

alengwenus, 2023-11-30:
Hi @nanosonde,
I pushed the files to this repo and added the necessary files for a custom_component.
Still missing some documentation but if you know how to add a custom_component to HomeAssistant (manual or as a user defined repo via HACS), the integration should be self explanatory.
Once the custom_component is added go to your integrations page and you should be able to add the "SMA EV Charger" integration which will setup all implemented entities.
Please provide feedback, wishes or issues!

nanosonde, 2023-11-30:
Hi @alengwenus,
thanks for the update.

I have already fiddled with it a bit.

This is how the config flow looks like in HA core 11.3.

[image]

However, it seems to be added correctly:

[image]

I wonder why the entities are editable.

Any ideas?

alengwenus, 2023-11-30:
Do not fiddle around. I can't support each fiddling. 😉
Just copy the custom_components folder from this repo into your config folder (where the configuration.yaml resides) and it should work just fine.

nanosonde, 2023-11-30:
> Do not fiddle around. I can't support each fiddling. 😉 Just copy the custom_components folder from this repo into your config folder (where the configuration.yaml resides) and it should work just fine.

Yes, that's what I finally did. See the result above. ;-)

alengwenus, 2023-11-30:
Looks right. 👍 I'll have a look at the config dialogue. On my system the labels are shown. Strange...

nanosonde, 2023-11-30:
Tested things so far:
* Operating mode: manually toggling between boost and stop -> working
* Session energy -> working
* Session status -> working shows "active" or "sleeping" if car is connected
* Station meter total -> working, shows the the same value as in the app
* Station power -> working,  shows the the same value as in the app
* Position rotary -> shows the correct value for boost (others not tested yet)
* LEDs status -> shows correctly, control not tested yet
* Disconnect after full charge -> shows correct status, control not tested yet
* Standby time -> shows correct value as in the parameters as well, control not tested yet

Things unclear (what do they do?): 
* Manual charging release
* Duration of the session (why can I control them?)
* Energy of the session  (why can I control them?)

closed 2023-12-01 by alengwenus