alengwenus/ha-sma-ev-charger #2: Request to add additional features

Xrlwyzz, 2023-12-01:
I just installed it and everything seems to be running great. Many thanks for this integration, that is a lot better than building everything manually!

The first ideas for improvement came up already:
Can you please also add the sensor for "Status of connected vehicle" `Measurement.Operation.EVeh.Health`? That would be great to use for some automations!
And also, the parameter "Initiate device restart" `Parameter.Sys.DevRstr` might be helpful sometimes.

alengwenus, 2023-12-01:
I added the vehicle status and the charging station status in 1.0.5.
Restarting the device would need a service call. But why would we ever have to restart the device?

Xrlwyzz, 2023-12-02:
Well, unfortunately I have to restart it fairly often. After a while, the EV Charger has hundreds of communication errors a day, which sometimes also stops charging for a few minutes (not shifting between one and three phases). 
Nobody found the error so far, but restarting helps…
There might be other reasons for a restart as well.

alengwenus, 2023-12-02:
I have to think about how to implement the restart function properly. The problem is, that if the device is restarted, the integration has to re-authenticate. At the moment this would mean that the integration needs to be reloaded. That's not easy to automize unless you completely restart HomeAssistant.
Let's see what's possible.

Xrlwyzz, 2023-12-04:
As long as that is only my request and nobody else needs it, please do not waste your energy on this issue. I did not expect it to be more compliacated than all the other settings. But your explanation makes sense.
I can help myself by just flicking the fuse as well, like I did before. Not everything has to be automated :-)

alengwenus, 2023-12-05:
I implemented a device action to restart the device. Re-authentication should also work as intended. It needs some re-work as I am not very satisfied with the current implementation. But it should work.
The new release is 1.0.6.

You can trigger the action in any automation by selecting "device" (in the actions tab), select your SMA EV Charger device and select the "Initiate restart" action.

alengwenus, 2023-12-07:
Thanks! Should be fixed in 1.0.7.

closed 2023-12-09 by alengwenus