alengwenus/ha-sma-ev-charger #3: Error in setup process (config flow)

adi-ba, 2023-12-10:
I have the following issue after intalling the HACS integration and trying to add the integration into home assistant.

[image]

Logger: homeassistant.config_entries
Source: config_entries.py:865
First occurred: 18:29:39 (1 occurrences)
Last logged: 18:29:39

Error occurred loading flow for integration smaev: cannot import name 'DeviceInfo' from 'homeassistant.helpers.device_registry' (/usr/src/homeassistant/homeassistant/helpers/device_registry.py)

alengwenus, 2023-12-11:
I guess you need to update your Home Assistant version. The DeviceInfo object was moved to device_registry.py in 2023.9. So you should not use any version below that for this integration to work.

adi-ba, 2023-12-11:
That was it! It works fine now. Thank you! :)

closed 2023-12-11 by adi-ba