alengwenus/ha-sma-ev-charger #10: Unknown status 5169 for channel Measurement.Operation.EVeh.ChaStt

jastBytes, 2023-12-28:
I recently discovered that sometimes the status for charging_session_status changes to 5169. Haven't found what this means and could not find anything on the internet or the python library defining those constants. As soon as I will see this again, I will look in the WebUI myself to see what it is.

[image]

alengwenus, 2023-12-28:
Would be great, if you can have a look at this. I checked my logs and the 5169 value never appeared in my setup.

jastBytes, 2023-12-28:
It is when the car is connected but the charger is locked. The german translation in the UI is "Ladestation gesperrt" the english version "Charging station locked". I would open an PR, but I am currently kind of sick. Will do if I find the time. 👍

alengwenus, 2023-12-29:
Thanks @jastBytes. I already implemented the session status value if the station is locked.

closed 2023-12-29 by alengwenus