alengwenus/ha-sma-ev-charger #12: sensors do not have the list of possible options

holgergrote, 2024-01-11:
The sensors

-schalterstellung_drehschalter
-status_ladevorgang
-status_des_verbundenen_fahrzeugs
-status_ladevorgang

have several options. The possible values are not listed under Entwicklerwerkzeuge->Zustände (development tools->states?). Other sensors do have an attribute options with all valid options.

This would be helpfull for automations. There it woud be possible to select the options from the GUI.
(Without that I had to search the correct value and type it manually. So this ist just a nice enhancement)

closed 2024-01-26 by alengwenus

alengwenus, 2024-01-26:
Hi @holgergrote,
I added the options for all the sensors which can be enumerated. Thanks for the hint!

holgergrote, 2024-01-27:
Checked the new version. I get the selectable options in for the automations. So it works.

Thanks for the enhancement.

PR #18 (2024-01-26): Add options for enum sensors
Fixes: #12