alengwenus/ha-sma-ev-charger #36: Cannot connect to SMA EV Charger

GithubTilo, 2024-10-30:
I use the same credentials with the HA integration I use for WebUI access.

Local EV charger IP: 192.168.10.29
Name + password.

The WebUI Access work

If I enter this credential in the HA dialog and press "Send," all fields will be deleted. Error: "connection failed". I also tried all ways using the "ssl checkboxes".

Do I need allow an external access to the wallbox in the EV Charger WebGui?

Thanks...

Tilo

alengwenus, 2024-10-31:
You should not have to activate external access, as this is only necessary for the SunnyPortal service.
If you get a "failed to connect" message then something is wrong with your network settings. Are you sure the IP address is correct? Does your HA instance has access to the local network your SMA EV Charger is in?

GithubTilo, 2024-10-31:
There must be an other problem. As you can see (attached screenshot) the home assistant terminal ping reach the SMA ev charger. First I thought the SSL Checkbox is the reason. But also with an unchecked SSL Checkbox can the connection not be established. Do I may need a special IP Port attached to the IP address.Also all user credentials are copied from the web GUI to prevent an incorrect entryHowever thank you for your very quick answer. In worst case I can still use the SMA App.Greetings Tilo

z3us22, 2024-12-14:
Did you try out login and password when connecting to the IP via your browser? In my case I don't know the login & password for the eCharger - I only got that data for the Home Manager and the portal. Did you find a solution?

closed 2025-03-16 by alengwenus