alengwenus/ha-sma-ev-charger #38: Question: eCharger Support?

danielbierstedt, 2025-04-25:
Hello, got questions about this plugin: does this work with the newer hardware release of the sma eCharger?
Why does it need the installer account? I would feel better with some normal user account for read only.

Thanks!

alengwenus, 2025-04-25:
I don't know if the eCharger uses the same API as the EV Charger. But I guess it's worth a try.

You can use a user account instead of an installer account. But not all entities will be available. The "missing" entities will be shown in the logs.

danielbierstedt, 2025-04-27:
Looks good to me. Working with a normal user, not sure what I'm missing.

[image]

I'm missing 500w somewhere, need to find out where my calculation is wrong...

[image]
The list is longer, don't know how to pull a list from commandline.

[image]

alengwenus, 2025-04-27:
If remember correctly there are some control elements (like setting the power and current limitation) does not work with the user account.

Are you using the old EV Charger or the new eCharger?

danielbierstedt, 2025-05-01:
This is the new eCharger, not the older EVCharger.

https://www.sma.de/produkte/ladeloesungen-elektromobilitaet/sma-echarger

closed 2026-02-05 by alengwenus