alengwenus/ha-sma-ev-charger #39: Set values for Parameter.Inverter.WMax/In

manttimue, 2025-05-12:
Hi there, I really, love this integration! I am missing just one thing: I have to adapt the maximum W number every now and then manually (to make sure I am using the optimum balance of PV Power and what's available in my battery - without consuming any grid power). So I am wondering if it'd be possible to add an entity in which these values could be set individually. 

All the best, and keep on the great work 🙂👍

alengwenus, 2025-05-13:
Hi @manttimue,
you can already set that value.
But the corresponding entity is disabled by default. Just go to the SMA EV Charger device and look for the hidden number entities. There you can select and activate it.
Note that you might need an installer account for this!

manttimue, 2025-05-16:
Thanks @alengwenus you were right, I found the settings afterwards and successfully added it to my dashboard, which works really fine.

Another question among us EV Charger users: Do you happen to know why the minimum power is set for 4500 but cannot be set below that? Or do you know which value needs to be adapted to lower the minimum range?