alengwenus/ha-sma-ev-charger #41: Charging session energy is showing Charging station meter reading in HA energy dashboard

UffeBj, 2025-07-07:
Hello,
I have seen a few times that in the HA energy dashboard, where I have defined the sensor.cargador_charging_session_energy as an individual device. That instead of showing the session energy, it shows the metering total, i.e the value of the sensor.smaev_3013045952_charging_station_meter_reading. So I have the total, forever, metering count displayed. (I hope the screenshot got attached).
Today I had a power outage at home, around 11.34 AM, affecting the EV charger, but not home assistant(UPS). Of course the SMA Home Manager 2.0 went down and also a switch between SHM2.0 and the EV charger. After the outage the EV charger, for whatever reason, does not recover connection to the meter, so I did a power cycle for the charger, which solves the issue. About 15min later a reading of 9,917125 MWh is recorded.

I do not know if the EV charger temporarily during is having the wrong values. But as soon as it is up I can see correct values in its web interface.

If more data or information is needed to better understand the issue, I am happy to assist.

Best regards, Ulf

[image]

[image]

alengwenus, 2025-07-28:
Hi @UffeBj,
I know this behavior. After a power outage of the EV Charger the parameter `Measurement.ChaSess.WhIn` shows the value of the parameter `Measurement.Metering.GridMs.TotWhIn.ChaSta`. When you reconnect the charging station to a car, the correct value will be displayed again. So I think this is not an issue with the integration but with the EV Charger itself.
However, for whatever reason, I am currently unable to reproduce the behavior. Maybe there was an update of the wallbox?

UffeBj, 2025-07-28:
Thanks for getting back! Good to know the issue has been seen before. I have the 1.2.23.R firmware, which I understand is the latest, even though it has been out for awhile. I will reconnect the cable the next time and check the behavior. Thanks again for your help.